SlimeVR/SlimeVR-Server #577: SlimeVR 1.0 Release

TheButlah, 2023-02-12:
This is a tracking issue for us to discuss and track updates on features, bugs, and UX that we want in a SlimeVR 1.0 release. Our goal of this release is to have the software "shippable". We would be confident that we did everything we needed for reviewers and users to be happy, especially by nontechnical users.

Issues remaining for 1.0:
* Do a polish pass on sensible defaults for features
* Do a ux polish pass to clean any warts
* Software autoupdates (or prompts for updates) so users will not get out of date due to laziness.
* Autobone UX done
* Mounting reset UX done
* Overlay uses file logging
* Overlay doesn't open a terminal
* Overlay starts with openvr
* Overlay is always aligned to first person perspective
* Installer installs overlay
* Overlay runs automatically in SteamVR
* GUI doesn't show overlay settings unless overlay is running
* Feeder app doesn't open a terminal
* OSC can use legtweaks and the other goodies that SteamVR has
* SteamVR latency is as good as OSC
* OpenVR bindings for full and quick reset active by default, without user intervention necessary.
* Config format should not serialize the default settings out, because it prevents those values from changing later when we change defaults.
* Rename everyone's old configs to a backup, and have one config to rule them all going forward, that handles forward compatibility properly.

Louka3000, 2023-02-12:
> OSC can use legtweaks and the other goodies that SteamVR has
Can just say "Support OSCQuery for VRChat"

ButterscotchV, 2023-02-12:
> Autobone UX done

AutoBone GUI needs a bit of a rework from its current state as in issue #446 and #283. Basically just needs to show the movements all before starting the recording and needs a proper finished visual guide for the movements.

ButterscotchV, 2023-02-12:
I think these issues should also be included in 1.0 or clarified/updated to fit it:
- https://github.com/SlimeVR/SlimeVR-Server/issues/405
- https://github.com/SlimeVR/SlimeVR-Server/issues/75

TheButlah, 2023-02-12:
> I think these issues should also be included in 1.0 or clarified/updated to fit it:
> 
>     * [[Server] Upload firmware via OTA #405](https://github.com/SlimeVR/SlimeVR-Server/issues/405)
> 
>     * [[Server] Building firmware by server #75](https://github.com/SlimeVR/SlimeVR-Server/issues/75)

Eiren mentioned in #75 that its actually not necessary for 1.0, so I've updated the issue to clarify.

TheButlah, 2023-02-13:
Descoped OSCQuery due to conversations in #580

ImUrX, 2023-03-07:
is there a tracking issue for having the tracker handling more than one server in the network?

Eirenliel, 2023-03-07:
#622 is required for 1.0.

TheButlah, 2023-03-12:
We have so much that is blocking 1.0 ;(

Eirenliel, 2023-03-12:
> We have so much that is blocking 1.0 ;(

This is not how it works.1.0 is not a feature that we all wait when it can be implemented to be blocked. We work forward and we will get there eventually.

TheButlah, 2023-03-13:
IK, was making an observation that there is *a lot* that should be done before reviewers have this in hand. At the current pace of things, it seems unrealistic to assume everything on this list will happen before reviewers review.

Eirenliel, 2023-06-15:
I love how this has nothing that we *actually* need for 1.0 or have done for 1.0 😢

closed 2025-09-24 by loucass003