SlimeVR/SlimeVR-Server #579: [Server] Migrate from old config to a more robust system

TheButlah, 2023-02-12:
The config system has historically been very brittle. We have had many config related issues over time:

* Old configs cause breakage because we didn't maintain backward compatibility with certain old config edge cases
* We don't really know or understand exactly what these edge cases are. Its hard to reason about what exactly would break stuff.
* We serialize the default values to disk, meaning that when we deserialize, it looks like the user intentionally set that data. This prevents us from changing our default values to better values, since the old configs force the old value.

Some improvements were made in https://github.com/SlimeVR/SlimeVR-Server/pull/229 but I think that the system is still quite brittle.

I would like to advocate for switching to a new config file format that we maintain forward and backward compatibility for. We can do this by:
* Indicating the version number of the config file in a field. If we ever make a backward-incompatible change, we bump that version number. The first version will be "1.0".
* Move all old config files lacking this version number to `oldconfig.backup`. This will reset users' configs. We will make no attempt to upgrade to the new 1.0 version, because honestly there are a lot of cursed configs out there and we don't want to risk further errors. Better to have pre-1.0 users redo their config than possibly crash or override new defaults. The only exception to this is the skeletal settings - we should try to import the skeletal settings to the new version. This is especially critical for early reviewers like Thrillseeker who likely haven't used it since then.

@ButterscotchV @loucass003 do y'all have any opinions on how to go about this?

Also, not to bikeshed but I heard that there were many, many issues with the yaml approach. I am down with json or toml as a config format (I know I said otherwise before, but I think I was proven wrong).

ButterscotchV, 2023-02-12:
In our current system, we do already have a version number `modelVersion`, and conversion between version numbers in `CurrentVRConfigConverter.java`, but as #266 states, we should handle any errors with loading and conversion better.

> We serialize the default values to disk, meaning that when we deserialize, it looks like the user intentionally set that data. This prevents us from changing our default values to better values, since the old configs force the old value.

This was planned with the version system, but I agree that it's a pain to have to do, especially for something like AutoBone, where the default values are changing quite often.

marcozzxx810, 2023-02-12:
I agree with the model version.

For tackling the problem of random input, internally we also need to build some validator for the config. 

I found the pros and cons for different formats: https://github.com/crystal-lang/shards/issues/25

I am more down to JSON format as it is simple and we do not need to deal wif special character