SlimeVR/SlimeVR-Server #580: [Server] Support VRChat OSCQuery

TheButlah, 2023-02-12:
To get feature parity with the steamvr driver, we need to know the position of the headset. We can receive this data from vrchat via OSCQuery.

TheButlah, 2023-02-12:
@Louka3000 if you have bandwidth to tackle this, can you assign yourself?

Louka3000, 2023-02-12:
I can add it to my todo, but it's not in high-priority.

TheButlah, 2023-02-12:
I think its actually very important, because steamvr is so much laggier than osc. Also reviewers are sure to show footage of it working on quest, and without legtweaks this footage won't be as good.

Eirenliel, 2023-02-12:
Also autobone doesn't work without it

Louka3000, 2023-02-12:
Autobone won't work simply by adding OSCQuery support, as we can't do Reset All (unless we implement manual user height in GUI, which we should, but is unrelated then).
Also, VRChat did not release any OSCQuery beta, so reviewers will be stuck with footage without HMD data.
Finally, even with OSCQuery, there's no way to get HMD rotation, so we'll be stuck with no neck solving... better give up feature parity and accept compromises :p

TheButlah, 2023-02-13:
> so we'll be stuck with no neck solving

Can you elaborate?

Louka3000, 2023-02-13:
1. OSCQuery won't give us any additional data compared to what you're getting by using OSC Trackers on a PC without the SteamVR driver. That means you ONLY get HMD height in % from the ground (0-1). There's no way to get height from VRChat.
2. As I said, implementing OSCQuery immediately won't work, as they didn't release it outside of closed beta.
3. There's no way to get HMD rotation, which means rotating your head will induce sliding, as if your Head shift and Neck length were 0. This is the current behaviour when using OSC without HMD data. The only workaround for this is to use a Slime tracker on your head.

TheButlah, 2023-02-13:
ok, thanks for the clarification it makes sense. I descoped this from 1.0.

@Louka3000 Does legtweaks work when using OSC on PCVR with the steamvr driver enabled?

Louka3000, 2023-02-13:
Yes, but they're the "bounce issue" with legtweaks because of VRChat tracking space synchronization. Maybe need to not send HMD data all the time if we have position?

Louka3000, 2023-02-13:
To clarify, both work just fine themselves.
The actual problem concerns the trackers being dragged with the HMD in VRChat with OSC Trackers, probably due to some latency. I don't know the exact cause as I haven't looked into it, and I'm too busy doing 9000 other things.

ButterscotchV, 2024-09-08:
Completed by #958.

closed 2024-09-08 by ButterscotchV