SlimeVR/SlimeVR-Server #582: [Server] Foot mounting calib should work

TheButlah, 2023-02-12:
@ButterscotchV said that the mounting calibration doesn't work for feet right now. This is because there is no good foot position to stand in that can be one at the same time as the skiing pose (too hard to stand on tip toes or heels).

evg-zhabotinsky, 2023-05-17:
Wouldn't it work if you actually get on your knees? Haven't checked, when I had the IMUs backwards the thought didn't occur, and after fixing that I never needed to calibrate mounting for feet, they are very forgiving in that regard.

ButterscotchV, 2023-05-17:
> Wouldn't it work if you actually get on your knees? Haven't checked, when I had the IMUs backwards the thought didn't occur, and after fixing that I never needed to calibrate mounting for feet, they are very forgiving in that regard.

Yes that could work? Seems a tiny bit iffy though. Currently my best thought is either sitting with your legs out in front, laying down, or legs kicked out in a chair as the same option (anything to tilt the foot backwards)

Stermere, 2023-06-17:
Another option would be to do it one foot at a time so that it can still be done standing up, this comes with its own drawbacks as well.

closed 2025-09-24 by loucass003