SlimeVR/SlimeVR-Server #585: Bundle and publish server release when a tag is created

Eirenliel, 2023-02-12:
Right now almost everything is already gets built on tag creation, but there are few things that needs finalizing:

* [ ] Bundle windows build
* [ ] Create draft release when a tag is created
* [ ] Upload required artifacts to the release
* [ ] Make sure everything is named properly

marcozzxx810, 2023-02-13:
Can we automate the step of publishing the app to flathub

ImUrX, 2023-02-13:
We could, but the toolchain used is too young and prone to breaking.
This ends up requiring someone to maintain it, like most package managers :/

Eirenliel, 2023-02-15:
Other changes to CI:

* [ ] Run bundle only after tests
* [ ] Run tests only after build

Eirenliel, 2023-02-15:
* [ ] Do not run bundle unless manualy ran or when a tag is created, it takes ages for PRs to bundle even though we don't need it

ImUrX, 2023-02-18:
Tests and building already take 2 minutes each, having them in dependence would make it 4m to finish actions. I can make bundle job depend on tests and build though.

Eirenliel, 2023-02-18:
Well my idea was since test requires build it can re-use the build artifact? It would also be nice for stuff to fail faster, right now each test starts even when some other tests fail, but we don't really need it in most cases, since if one fails we see the problem. Making bundle depend on others is the minimum, yeah.

Also clean up artifacts generated too. Right now linux and mac are bundled in server action, but windows gui is built in gui action while windows server is build in server action...

ImUrX, 2023-02-19:
every OS binary is built on GUI action, I think I forgot to add a macOS artifact though.

I don't think the artifact reuse will work, because the server artifact is an uber jar one. Let's try without test needing for build for now

PR #599 (2023-02-22): Upload artifacts to draft when new tag
Closes #585 
Requires testing, will do tomorrow
closed 2023-02-22 by Eirenliel