SlimeVR/SlimeVR-Server #589: [GUI] add a censor toggle for IP

marcozzxx810, 2023-02-14:
The current GUI show the local IP and it may cause some privacy issue. Therefore we would like to censor it by default.

Proposal:
- show as ***.***.***.*** by default
- when the user clicks on the IP, the IP will reveal
- when the IP is revealed and the user clicks it again, the IP changes to ***.***.***.***

Eirenliel, 2023-02-14:
Not sure if it's actually a privacy issue, it's just local ip and it's only on the settings page

marcozzxx810, 2023-02-14:
@Eirenliel I agree but people without technical background may not fully understand the local ip and public ip. To remove confusion, we can just hide by default and user can click on it when they need SilmeVR ip to connect to.