SlimeVR/SlimeVR-Server #619: Intergrated serial console cannot recognise ESP32-C3 USB CDC port

TerayTech, 2023-03-03:
Hi, there~
I'm building a customized hardware slime tracker with ESP32-C3 Wifi Module. There is a USB Phy controller intergrated in the ESP32C3 and can be used as a serial interface or a JTAG debug interface.
When it pluged into the computer running win10 the serial port can be detected in the device manager, but in the serial console built in the slimevr software it doesn't appears.

TerayTech, 2023-03-03:
[image]

evg-zhabotinsky, 2023-05-17:
Just to make sure, it doesn't let you set the wifi passwort through the server either? (Probably so.)

The server has a whitelist of "related" serial ports, currently checked by "descriptive name" : https://github.com/SlimeVR/SlimeVR-Server/blob/d4068e7ccb6e6ba32aa8c09f4764c90fa5a0777a/server/src/main/java/dev/slimevr/serial/SerialHandler.java#L250-L265

It should be a matter of just adding the name to the list, but… If it really is the same name that device manager shows (I'm not sure), system language not being "English" may cause issues with that approach. Not sure what the best fix would be then, though removing the last line should help in this specific case and shouldn't cause too many extra issues.

evg-zhabotinsky, 2023-05-17:
Also probably related to #362 , if so it is waiting for the serial library fix too.
P.S. The name probably would be "USB Serial Device" in English, the exact same as for Index bt adapter.

nekomona, 2023-07-04:
It's a localization error, where integrated serials are labeled `USB 串行设备` instead of `USB Serial Device`, while the firmware checks for `usb` and `seri` at here:
https://github.com/SlimeVR/SlimeVR-Server/blob/d4068e7ccb6e6ba32aa8c09f4764c90fa5a0777a/server/src/main/java/dev/slimevr/serial/SerialHandler.java#L257-L265

Could remove the `&&` statement on L264 and only looks for `usb`, which would be a working one. Yet more sophisticated UART detection need to be implemented to address more similar issues.

TheDevMinerTV, 2023-07-04:
If it were to detect all devices with the name containing `usb`, I'd have a somewhat big issue since almost every device that I use includes `USB` in it's name

Eirenliel, 2023-07-04:
we need to use VID/PID, which should be possible after #708

TheDevMinerTV, 2023-07-04:
> Yet more sophisticated UART detection need to be implemented to address more similar issues.

Can't we just look for `(COM*)` in the name for now?

Eirenliel, 2023-07-04:
that will definetely be more positives, this only works for serial ports already, so they all have COM in them and/or USB, so grabbing all of them is not a good idea. we need VID/PID

ImUrX, 2023-11-05:
this probably is fixed by #756 which added a whitelist.

do open it again if it's not

closed 2023-11-05 by ImUrX