SlimeVR/SlimeVR-Server #623: [GUI] Improve tracker page

Eirenliel, 2023-03-07:
* [ ] If mounting reset was used the mounting should show as "Custom"
  * [ ] Choosing other mounting should change it to be hand-set again
* [ ] Tracker 3D view should switch between raw and calibrated the same way the rotation on the home screen does
* [ ] 3D model of the tracker should be updated to something more resembling slime model so people knew better where everything is, right now it's even long instead of tall so it's doubly confusing
* [ ] Tracker 3D view should be enabled by default and switching it on or off should presist between opening tracker pages

marcozzxx810, 2023-03-11:
>  If the mounting reset was used the mounting should show as "Custom", Choosing other mounting should change it to be hand-set again

Do you mean the mounting position should show Custom after triggering the mounding reset? also what It means by hand-set

> tracker 3D view should switch between raw and calibrated the same way the rotation on the home screen does

We can add a toggle on top of the 3D view for raw and calibrated, I wonder if we need to persist the toggle setting here. 

> 3D model of the tracker should be updated to something more resembling slime model so people knew better where everything is, right now it's even long instead of tall so it's doubly confusing

Three.js suggested to use a GLB / .GLTF 3D model, also support FBX、OBJ, COLLADA

> Tracker 3D view should be enabled by default and switching it on or off should presist between opening tracker pages

we can have the 3D view toggle setting placed in the general setting? actually, why does the user need to turn it off? they can just close the page.

Eirenliel, 2023-03-23:
* [ ] Tracker should have a warning sign if it was reconnected and wasn't fully reset. The warning should also be on the home page.

Eirenliel, 2023-04-14:
* [ ] Treacher should have a warning sign if there is more than one tracker assigned to the same body part. The warning should also be on the home page.

closed 2025-03-04 by Eirenliel