SlimeVR/SlimeVR-Server #625: New user experience notes

Eirenliel, 2023-03-07:
In this issue we can collect all the notes on what and how we should communicate something to new users. This doesn't have to be solved in one specific way, we can discuss and find better ways. This might be even done in docs, but ideally we want users to not need docs. This issue is so we don't forget.

* [ ] Tracking quality depends on body proportions a lot
* [ ] You need to either set mounting correctly or run mounting reset each time (preferably). Maybe we should even show warning that mounting is not set or if the tracker/server is restarted but mounting reset wasn't done (and mounting isn't set manually) it should warn the user about that problem.
* [ ] We should suggest mounting positions and directions

Please feel free to add your suggestions in the comments.

marcozzxx810, 2023-03-11:
> You need to either set mounting correctly or run mounting reset each time (preferably). Maybe we should even show warning that mounting is not set or if the tracker/server is restarted but mounting reset wasn't done (and mounting isn't set manually) it should warn the user about that problem.

can show a warning icon beside the tracker, then the user hovers over it can see mounting is not set. 

Actually, if the user has passed the onboarding process, I think we better have a snack bar show device is connected, and redirect the user to the tracker setting page. ( not sure is this a good idea tough)

loucass003, 2025-09-24:
What do we do about this .....