SlimeVR/SlimeVR-Server #626: [Feature Request] Send Battery percentage from Server software to SteamVR

OzzyHelix, 2023-03-09:
I was wondering if it would be possible to send the battery info from trackers to SteamVR maybe via the server I'm not sure how it would be implemented but it would be helpful to see the battery status of trackers within SteamVR

marcozzxx810, 2023-03-11:
Seem we need to update this three props: Prop_DeviceProvidesBatteryStatus_Bool, Prop_DeviceBatteryPercentage_Float, Prop_DeviceIsCharging_Bool in driver repo

Ref: https://github.com/ValveSoftware/openvr/blob/8f71cf8d9892006939f5e7a9473b6984387a8ec3/docs/Driver_API_Documentation.md#device-properties

OzzyHelix, 2023-03-11:
the reason I suggested grabbing the battery percentage from the server. I assumed that might be the easiest way to send SteamVR battery info. I don't know how the SteamVR drivers and server fully work so my assumption could be completely wrong

ImUrX, 2023-07-26:
This was done by #668

closed 2023-07-26 by ImUrX