SlimeVR/SlimeVR-Server #631: The VRServer is not gracefully shutdown

marcozzxx810, 2023-03-14:
How to reproduce:
1. add a `logManager.info("hello") in https://github.com/SlimeVR/SlimeVR-Server/blob/main/server/src/main/java/dev/slimevr/VRServer.java#L179

2. ./gradlew shadowJar

3. put the `slimevr.jar` from `SlimeVR-Server\server\build\libs` to the your installation file

4. The log doest has hello.

My proposal is that:

1. Tauri should send sigterm to the server
2. Then server should start the clean up process

Obstacle:
1. now the server cannot handle SIGINT, need to investigate how to do so

Already tried to put a shutdown hook in the main function, but It still cannot work. Not sure it related to gradle

marcozzxx810, 2023-03-14:
This also create a problem of log lock is still there when the server close

ImUrX, 2023-07-26:
This was fixed by #632

closed 2023-07-26 by ImUrX