SlimeVR/SlimeVR-Server #654: Default orientation for some trackers

Eirenliel, 2023-04-05:
Default mounting orientations for some trackers should be set according to our recommendations, not just front. Currently it's: left upper arm, left lower arm, left hand - left; right upper arm, right lower arm, right hand - right; hip - right.

Louka3000, 2023-04-05:
Chest, waist and hip should be front! Not side!
Thighs: front
Ankles: side (outwards)

Eirenliel, 2023-04-05:
Well The Docs™ recommends waist on the side: https://docs.slimevr.dev/server/putting-on-trackers.html. Which I belive is the leftower from when it was factually hip... We should get to a consensus, recommend this to users and use as a default, since Poly and Uriel are designing the page in wizard about putting trackers on.

Louka3000, 2023-04-05:
That’s just your picture that I didn’t remove. We could remove them now.
[image]

Eirenliel, 2023-04-05:
Okay. Then front. Thought it still says "or left", but yea...

Eirenliel, 2023-06-15:
> #679 should be completed for this

Why? No.

Louka3000, 2023-06-15:
> > #679 should be completed for this
> 
> Why? No.

So that ankles and upper arms can be diagonal

Eirenliel, 2023-06-15:
We just need them to be on the side by default it's way better than front already. Please don't block this feature by another feature (that I honestly don't think should exist at all). We need something that is better than current one. If we add diagonal mounting in the future (I don't think we should), we can change defaults.

PR #898 (2023-11-19): Add default mounting orientation for certain body parts
Fixes #654
closed 2023-11-19 by ImUrX