SlimeVR/SlimeVR-Server #669: Dead battery doesn't show up

Eirenliel, 2023-04-11:
The tracker with low battery (blinking constantly but still connected) doesn't show battery indicator in the server GUI. If put on charge it will show 100% and charging, but when unplugged the battery icon will just disappear along with percentage and voltage. Not sure if it's GUI, Server or firmware problem...

sctanf, 2023-04-12:
Having a similar issue where a device with 0% battery but still active will not show the battery in gui
I think the tracker should explicitly say it does not have a battery rather than the server/gui guessing.. same for charge status, since it is checking for a high voltage near 5V

Eirenliel, 2023-04-12:
The tracker sends battery level either way, the problem is in the gui or server.

sctanf, 2025-04-25:
I think while charging it may be a good idea to replace the battery charging icon with just a charging indicator as some more generic way to indicate plugged in, and then hide the battery level

When the tracker is at 0% battery, the gui could keep showing the battery, and only hide the battery completely if the voltage was out of range
nRF trackers considers <1.5V or >6V as a battery missing condition

PR #1410 (2025-04-28): Better battery and wifi statuses behavior
Battery will continue to show even if the level is 0%, and the level will be hidden if in a charge state. Ping will be hidden if it is null. Firmware updater will also treat 0% level as low battery.

Fix #669 and #1112
closed 2025-04-28 by loucass003