SlimeVR/SlimeVR-Server #674: Connection lost to the server. Trying to reconnect...

xkeyC, 2023-04-16:
###  My SlimeVR-Server Always Trying to reconnect

### Logs
2023-04-16 14:55:12 [INFO] Running version v0.7.1
2023-04-16 14:55:12 [INFO] Starting the tracker server on port 6969...
2023-04-16 14:55:13 [INFO] [Keybinding] Bound full reset to CTRL+ALT+SHIFT+Y
2023-04-16 14:55:13 [INFO] [Keybinding] Bound yaw reset to CTRL+ALT+SHIFT+U
2023-04-16 14:55:13 [INFO] [Keybinding] Bound reset mounting to CTRL+ALT+SHIFT+I
2023-04-16 14:55:13 [INFO] [SteamVR Driver Bridge] Pipe \\.\pipe\SlimeVRDriver created
2023-04-16 14:55:13 [INFO] [SteamVR Feeder Bridge] Pipe \\.\pipe\SlimeVRInput created
2023-04-16 14:55:13 [INFO] [SteamVR Driver Bridge] Pipes are created
2023-04-16 14:55:13 [INFO] [SteamVR Feeder Bridge] Pipes are created
2023-04-16 14:55:13 [INFO] [WebSocket] Web Socket VR Bridge started on port 21110
2023-04-16 14:55:13 [INFO] [WebSocketAPI] New connection from: 0:0:0:0:0:0:0:1
2023-04-16 14:55:15 [INFO] [WebSocketAPI] Disconnected: 0:0:0:0:0:0:0:1, (1006) The connection was closed because the other endpoint did not respond with a pong in time. For more information check: https://github.com/TooTallNate/Java-WebSocket/wiki/Lost-connection-detection. Remote: false
2023-04-16 14:55:18 [INFO] [WebSocketAPI] New connection from: 0:0:0:0:0:0:0:1
2023-04-16 14:55:20 [INFO] [WebSocketAPI] Disconnected: 0:0:0:0:0:0:0:1, (1006) The connection was closed because the other endpoint did not respond with a pong in time. For more information check: https://github.com/TooTallNate/Java-WebSocket/wiki/Lost-connection-detection. Remote: false
2023-04-16 14:55:23 [INFO] [WebSocketAPI] New connection from: 0:0:0:0:0:0:0:1
2023-04-16 14:55:25 [INFO] [WebSocketAPI] Disconnected: 0:0:0:0:0:0:0:1, (1006) The connection was closed because the other endpoint did not respond with a pong in time. For more information check: https://github.com/TooTallNate/Java-WebSocket/wiki/Lost-connection-detection. Remote: false
2023-04-16 14:55:28 [INFO] [WebSocketAPI] New connection from: 0:0:0:0:0:0:0:1
2023-04-16 14:55:30 [INFO] [WebSocketAPI] Disconnected: 0:0:0:0:0:0:0:1, (1006) The connection was closed because the other endpoint did not respond with a pong in time. For more information check: https://github.com/TooTallNate/Java-WebSocket/wiki/Lost-connection-detection. Remote: false
2023-04-16 14:55:33 [INFO] [WebSocketAPI] New connection from: 0:0:0:0:0:0:0:1
2023-04-16 14:55:35 [INFO] [WebSocketAPI] Disconnected: 0:0:0:0:0:0:0:1, (1006) The connection was closed because the other endpoint did not respond with a pong in time. For more information check: https://github.com/TooTallNate/Java-WebSocket/wiki/Lost-connection-detection. Remote: false
2023-04-16 14:55:38 [INFO] [WebSocketAPI] New connection from: 0:0:0:0:0:0:0:1
2023-04-16 14:55:40 [INFO] [WebSocketAPI] Disconnected: 0:0:0:0:0:0:0:1, (1006) The connection was closed because the other endpoint did not respond with a pong in time. For more information check: https://github.com/TooTallNate/Java-WebSocket/wiki/Lost-connection-detection. Remote: false
2023-04-16 14:55:41 [INFO] VRServer thread interrupted

https://user-images.githubusercontent.com/39891083/232278621-510ec8b5-d667-4883-8a89-b15f47ce60c1.mp4

xkeyC, 2023-05-12:
Disable windows11 bbr2 can fix it

```
netsh int tcp set supplemental template=internet congestionprovider=CUBIC
netsh int tcp set supplemental template=internetcustom congestionprovider=CUBIC
netsh int tcp set supplemental template=Datacenter congestionprovider=CUBIC
netsh int tcp set supplemental template=Datacentercustom congestionprovider=CUBIC
```

This may be a problem with upstream TooTallNate/Java-WebSocket, or it may be a Microsoft's bug.

And enable bbr for reproduce the bug:
```
netsh int tcp set supplemental Template=Internet CongestionProvider=bbr2
netsh int tcp set supplemental Template=Datacenter CongestionProvider=bbr2
netsh int tcp set supplemental Template=Compat CongestionProvider=bbr2
netsh int tcp set supplemental Template=DatacenterCustom CongestionProvider=bbr2
netsh int tcp set supplemental Template=InternetCustom CongestionProvider=bbr2
```

ImUrX, 2023-11-05:
This probably is fixed by https://github.com/meetrevision/playbook/issues/27

closed 2023-11-05 by ImUrX