SlimeVR/SlimeVR-Server #680: Drift compensation doesn't consider jump from 180 to -180

CreeperCrafter979, 2023-04-19:
Using owoTrack on a tablet with imperceptible drift without the magnetometer and setting drift compensation settings to 100% using the previous 6 resets, I've run the following test:
Set the tracker on a completely still and flat surface, clear drift compensation and reset, wait at least 30 seconds, spin the tracker in a full 360 degrees spin, and reset a final time.
This should theoretically cause SlimeVR to immediately start overcompensating drift, but it remains completely static as the raw Y rotation has gone from 0 to 180, flip to -180, and back to 0. When drift is calculated, the server seems to subtract the previous raw Y rotation from the current raw Y rotation which here results in 0, no drift. If you spin the tracker by 350 degrees instead of a full circle, the drift will be calculated as -10 when it should be 350.

This issue will be a headache to deal with when actually using the trackers for virtual reality or motion capture proved with the following test:
Set the tracker on a completely still and flat surface with its raw Y rotation at 0 degrees, clear drift compensation and reset, rotate the tracker 90 degrees, wait at least 30 seconds, reset again, spin the tracker at a speed such that the displayed Y rotation stays near 0 degrees the entire time, reset once the tracker's raw Y rotation is back to 90 degrees.
This will cause drift compensation to slow down and become misaligned with the tracker that's rotating at a constant speed. When using the trackers normally, this could appear as the drift being calculated correctly after the first two resets but spiraling out of control once you reset a third time. 

I originally did these tests on a Linux computer using Java, but the first test also proved faulty drift calculation on a Windows laptop using the executable. Latest SlimeVR-Server 0.7.1 on both.

Possible fixes:
1. Keep note of how many times each tracker's raw Y rotation jumps between 180 to -180 degrees and calculate drift with this accounted for.
2. Stop the raw Y rotation from jumping and instead let it continue outside the -180 to 180 degree bounds.
3. (Preferred by me) Don't calculate drift between resets. Instead, add tracker drift calibration as another part of setup or as a new button in each tracker's settings menu. Simply put each tracker on a still surface, click a button or switch, and the drift calculated in this duration of time could be saved to the vrconfig.yml file for future and all use.

Louka3000, 2023-04-19:
Thanks for the report. This could indeed affect users if they happen to wait for their tracker to have drifted more than 180 degrees before resetting… which is pretty unlikely however.

Proposed fixes 1 and 2 are the same (because of the way Quaternions work, we indeed have to rely on the fix 1); fix 3 is unrelated to our dynamic drift calculation and is already handled IMUs’ boot-up calibrations.

Mattshark89, 2023-09-07:
I'm having an issue I believe is related to this one... Over time, drift compensation becomes absolutely useless and I get a constant drift that resetting never seems to reduce. The only fix for me is to restart the SlimeVR server.
My suspected reason is perhaps the sensor's raw (raw, not recalculated based on drift) rotation has finally made the 180/-180 flip, causing this round-the-world issue to arise.

Also: if the user does their first reset at a raw -179 degrees, then the imu drifts two degrees (raw 179), what happens then? Are you screwed from the get-go?

Louka3000, 2023-09-07:
> Also: if the user does their first reset at a raw -179 degrees, then the imu drifts two degrees (raw 179) by the second reset, what happens then? Is that sensor screwed from the get-go?
No, it uses the adjusted rotation.

loucass003, 2025-09-24:
Drift compensation is gone. Closing

closed 2025-09-24 by loucass003