SlimeVR/SlimeVR-Server #686: Battery voltage is inaccurate / Battery percentage should be done in Server

NWBx01, 2023-04-26:
Due to resistor tolerancing, the reported battery voltage in the server is inaccurate. This isn't the biggest issue, but it causes problems down the line with [reporting battery info to SteamVR](https://github.com/SlimeVR/SlimeVR-Server/pull/668). For example, the highest a tracker's battery level should go is approximately 4.2V, as this is what the TP4056 maxes out at for constant current charging. However, it's not uncommon to see users with reported battery levels beyond 4.2V.

Therefore, in my opinion, battery percentage should really be calculated on the server side rather than on the tracker. The main reasoning for this would be to apply a scaling factor so that erroneous readings above 4.2V get scaled to 4.2V maximum, with 4.15V treated as 100% battery. As you might be able to intuit, if the voltage is incorrect on the high end, it will also be incorrect on the low end. Again, this causes issues with [reporting battery info to SteamVR](https://github.com/SlimeVR/SlimeVR-Server/pull/668). Namely, if the battery voltage is reported to be higher than it truly is, it may not meet the threshold of <15% battery percentage that SteamVR uses to determine if a tracker is low battery life, or otherwise this may occur later than it should (perhaps when the battery is realistically much lower than 15%, more like 10%, the user believes their tracker to be at 15%). In theory, scaling from nearer to 0% would be possible, but it is much more difficult to reliably tell; a user is much more likely to begin using their trackers after they charge to 100% (>=4.15V) than they are to drain their battery all the way to 0%, so trying to determine if what's reported as 5% is really 0% is much more difficult to reliably determine.

Likewise, the way the GUI determines a tracker to be charging is a simple check of >=4.3V. From what has already been mentioned, you should be able to understand why this unreasonable. At the very least it should be a check of >=4.15V, but preferably would be a check over time e.g. batteryVoltage1 > batteryVoltage2 -> isCharging = true, batteryVoltage1 <= batteryVoltage2 -> isCharging = false.

[image]

sctanf, 2023-04-26:
For determining maximum voltage reads, how can you tell between charging and battery voltage? The tracker will always read close to 5V when plugged in due to diodes, but resistor tolerances can blur the lines between high battery voltage and charging voltage. Even if the threshold between charging/not charging is large enough, battery voltage needs time to settle which can take a few minutes. This is why some voltages below 4.2V are considered 100% originally. The TP4056 also has its own tolerances and it wouldn't be uncommon to see large voltage errors there too.

I don't think the server should need to do more when it comes to battery calculation, something like this should be solved on the tracker. It can save battery data to flash storage, for example. Charging status could also be moved to the tracker instead, since a tracker might not read >4.3V but still be charging, or something like a phone cannot show as charging in the GUI without spoofing the voltage to above 4.3V

NWBx01, 2023-04-26:
> The tracker will always read close to 5V when plugged in due to diodes
Will it? The TP4056 should only output 4.2V near the end of the charging cycle. Reading close to 5V would mean that the TP4056 is just passing through the 5V off of the USB?

[image]

Eirenliel, 2023-04-26:
Server should be hardware agnostic and not assume anything about the voltage. If it gets percentage from the tracker, it has to use it. We should also add tracker reporting charging startus itself. If you want to add any scaling, it should be done in firmware.

sctanf, 2023-04-26:
The TP4056 is incapable of passing 5V, it has no power path functionality. This is integrated separately by SlimeVR Trackers, and on devices with no diodes, this function doesn't exist.

So most trackers will read close to 5V plugged in, and some will just read battery voltage

NWBx01, 2023-04-26:
> So most trackers will read close to 5V plugged in, and some will just read battery voltage

I have genuinely never experienced this which is why I asked. I'll have to take your word for it since mine always topped off at 4.2V with diodes.

sctanf, 2023-04-26:
> > So most trackers will read close to 5V plugged in, and some will just read battery voltage
> 
> I have genuinely never experienced this which is why I asked. I'll have to take your word for it since mine always topped off at 4.2V with diodes.

Ahh my mistake, batt read is off the battery directly in diy schematic