SlimeVR/SlimeVR-Server #726: Tap reset noises can play during onboarding

Vyolex, 2023-06-14:
When first setting up slimes, once they are assigned to body parts the tap reset noises can be "accidentally" triggered, this is very confusing when we are not yet introduced to their function. It might be nice if we can supress this until we get past the reset tutorial. (even if just muting the sound) 

This is only really an issue on the body assignment page as that is the only page it can be triggered before the reset tutorial is presented.

Louka3000, 2023-06-14:
Body assignment needs tap since it can use it to assign (it is very convenient)

Vyolex, 2023-06-14:
Those taps are independent afaik, can correct me if I'm wrong.

ImUrX, 2023-06-14:
this is a state nightmare but to fix it you just need to enable setup mode on onboarding and disable it when doing the tutorial