SlimeVR/SlimeVR-Server #764: Use path alias instead of relative path in GUI project

marcozzxx810, 2023-07-04:
Using a relative path seems hard to be understood, For example:

https://github.com/marcozzxx810/SlimeVR-Server/blob/main/gui/src/components/onboarding/pages/ConnectTracker.tsx#L12

Proposed solution

- [ ] use `@` to indicate root path for GUI

Current
```ts
import { useLayout } from '../../../hooks/layout';
import { useOnboarding } from '../../../hooks/onboarding';
```
Later
```ts
import { useLayout } from '@/hooks/layout';
import { useOnboarding } from '@/hooks/onboarding';
```

ImUrX, 2023-07-04:
you mean `@hooks/` i guess, does the editor do this when auto-importing a new module or does it still use relative pathing? Can we enforce this through eslint?

marcozzxx810, 2023-07-04:
I mean '@/hooks' so we have '@' pointing to gui folder. It depends on your LSP but editor with intellisense should be able to pick up tsconfig

For eslint, https://www.npmjs.com/package/eslint-import-resolver-typescript  I am not so certain about can we enforce it

ImUrX, 2023-07-04:
i meant, auto-imports will prefer using the aliases or relative paths?

marcozzxx810, 2023-07-04:
vscode will prefer the alias

TheDevMinerTV, 2023-07-04:
> vscode will prefer the alias

It depends on what the user configured, though we can override this by including a `.vscode` folder with the new settings.

ImUrX, 2023-07-04:
then its fine, sure :+1:

PR #771 (2023-09-21): feat: use path-alias
close: #764
closed 2023-09-21 by Louka3000