SlimeVR/SlimeVR-Server #819: Appimage is broken: libthai.so.0 not found

wtarreau, 2023-08-13:
Hello,

it's impossible to execute the AppImage for me, as you can see AppRun.wrapped depends on libthai.so.0 which is not shipped with the image (and obviously not present on my system, though it shouldn't have to):

```
$ ./SlimeVR-amd64.appimage 
slimevr: error while loading shared libraries: libthai.so.0: cannot open shared object file: No such file or directory
```
Even after --appimage-extract:
```
$ ./AppRun.wrapped 
slimevr: error while loading shared libraries: libthai.so.0: cannot open shared object file: No such file or directory
```
Strangely the binary doesn't depend on it, it's just extra stuff that it seems to be loading at run time:
```
$ ldd ./AppRun.wrapped 
        linux-vdso.so.1 (0x00007fffae564000)
        libc.so.6 => /lib64/libc.so.6 (0x00007ffab55fc000)
        /lib64/ld-linux-x86-64.so.2 (0x00007ffab5823000)
```
```
$ strace -e trace=openat ./AppRun.wrapped
...
openat(AT_FDCWD, "tls/x86_64/x86_64/libthai.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "tls/x86_64/libthai.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "tls/x86_64/libthai.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "tls/libthai.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "x86_64/x86_64/libthai.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "x86_64/libthai.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "x86_64/libthai.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
...
```
etc.

I think there's either an undesired dependency on your system or one that was not identified at packaging time.

ImUrX, 2023-08-13:
[libthai is excluded from appimages](https://github.com/AppImageCommunity/pkg2appimage/blob/master/excludelist#L120), they are version based apparently.
Current appimages also fail when trying to use it without GStreamer

wtarreau, 2023-08-13:
Ah OK thanks for the info. So basically AppImages are no longer portable images, and they heavily depend on the target system, then they've already become pointless. The initial idea was good regardless, too bad they shredded it. One alternative might be to make sure not to depend on that libthai that I had never heard of.

ImUrX, 2023-08-13:
I don't know tbh, I should consult because we don't actually pack the appimage, will try to figure it out.

The ldd output of yours as you can see is just a wrapper binary. This is the ldd of the actual binary: 
```
ldd target/release/slimevr
        linux-vdso.so.1 (0x00007ffd40895000)
        libwebkit2gtk-4.0.so.37 => /nix/store/cklqv2zvzcb82zxn1ig0gs7xaqc0n7n0-webkitgtk-2.40.2+abi=4.0/lib/libwebkit2gtk-4.0.so.37 (0x00007f261bc00000)
        libgtk-3.so.0 => /nix/store/kx164f6plmdd1d1nl6k3nkid6jh9bsm8-gtk+3-3.24.37/lib/libgtk-3.so.0 (0x00007f261b400000)
        libgdk-3.so.0 => /nix/store/kx164f6plmdd1d1nl6k3nkid6jh9bsm8-gtk+3-3.24.37/lib/libgdk-3.so.0 (0x00007f262207c000)
        libcairo.so.2 => /nix/store/67lcda9pla0kmqzabpl3nxxq7z6v4rr6-cairo-1.16.0/lib/libcairo.so.2 (0x00007f26206c4000)
        libgdk_pixbuf-2.0.so.0 => /nix/store/0dlkl57qg0l063d4zdcrnw4pyrnzgf60-gdk-pixbuf-2.42.10/lib/libgdk_pixbuf-2.0.so.0 (0x00007f262204f000)
        libsoup-2.4.so.1 => /nix/store/gibp8yli4p6ylqyvqpf7k2vd803iwckp-libsoup-2.74.3/lib/libsoup-2.4.so.1 (0x00007f2620623000)
        libgio-2.0.so.0 => /nix/store/sxzpfcldisjqlgq8mm7435yz4bmbaalb-glib-2.76.2/lib/libgio-2.0.so.0 (0x00007f261b215000)
        libjavascriptcoregtk-4.0.so.18 => /nix/store/cklqv2zvzcb82zxn1ig0gs7xaqc0n7n0-webkitgtk-2.40.2+abi=4.0/lib/libjavascriptcoregtk-4.0.so.18 (0x00007f2619600000)
        libgobject-2.0.so.0 => /nix/store/sxzpfcldisjqlgq8mm7435yz4bmbaalb-glib-2.76.2/lib/libgobject-2.0.so.0 (0x00007f261959f000)
        libglib-2.0.so.0 => /nix/store/sxzpfcldisjqlgq8mm7435yz4bmbaalb-glib-2.76.2/lib/libglib-2.0.so.0 (0x00007f2619458000)
        libgcc_s.so.1 => /nix/store/n7pvb7gdf1g6dvj7sl92i882qjl4kyx9-gcc-12.3.0-lib/lib/libgcc_s.so.1 (0x00007f2619437000)
        libm.so.6 => /nix/store/flf14c3ibr83jsa070j25hg5gjapydhl-glibc-2.37-8/lib/libm.so.6 (0x00007f2619357000)
        libc.so.6 => /nix/store/flf14c3ibr83jsa070j25hg5gjapydhl-glibc-2.37-8/lib/libc.so.6 (0x00007f2619171000)
        /nix/store/flf14c3ibr83jsa070j25hg5gjapydhl-glibc-2.37-8/lib/ld-linux-x86-64.so.2 => /nix/store/ibp4camsx1mlllwzh32yyqcq2r2xsy1a-glibc-2.37-8/lib64/ld-linux-x86-64.so.2 (0x00007f262218d000)
        libwpe-1.0.so.1 => /nix/store/4cl32da38qlhn8scsp2j39ind9rpsr79-libwpe-1.14.1/lib/libwpe-1.0.so.1 (0x00007f2620613000)
        libWPEBackend-fdo-1.0.so.1 => /nix/store/w53ikdp7jk2lj9y57kh3yxb0vvb0qcl8-wpebackend-fdo-1.14.2/lib/libWPEBackend-fdo-1.0.so.1 (0x00007f2619159000)
        libepoxy.so.0 => /nix/store/6mxy42imfq6klir4gs74s7hghap0v788-libepoxy-1.5.10/lib/libepoxy.so.0 (0x00007f2619024000)
        libatomic.so.1 => /nix/store/n7pvb7gdf1g6dvj7sl92i882qjl4kyx9-gcc-12.3.0-lib/lib/libatomic.so.1 (0x00007f2620608000)
        libicudata.so.73 => /nix/store/sbmlx3lfc9m12qka9hl1069hpwhfc9na-icu4c-73.1/lib/libicudata.so.73 (0x00007f2617000000)
        libicui18n.so.73 => /nix/store/sbmlx3lfc9m12qka9hl1069hpwhfc9na-icu4c-73.1/lib/libicui18n.so.73 (0x00007f2616c00000)
        libsystemd.so.0 => /nix/store/8pbr7x6wh765mg43zs0p70gsaavmbbh7-systemd-253.3/lib/libsystemd.so.0 (0x00007f2618f2e000)
        libicuuc.so.73 => /nix/store/sbmlx3lfc9m12qka9hl1069hpwhfc9na-icu4c-73.1/lib/libicuuc.so.73 (0x00007f2616800000)
        libz.so.1 => /nix/store/k54w79m79x2hkc1i8ps7l4fry576dpvs-zlib-1.2.13/lib/libz.so.1 (0x00007f2618f0f000)
        libpangocairo-1.0.so.0 => /nix/store/yvf9s71nkckxvr5p0szvim0cp43ybi9s-pango-1.50.14/lib/libpangocairo-1.0.so.0 (0x00007f261bbef000)
        libpango-1.0.so.0 => /nix/store/yvf9s71nkckxvr5p0szvim0cp43ybi9s-pango-1.50.14/lib/libpango-1.0.so.0 (0x00007f2618ea4000)
        libharfbuzz.so.0 => /nix/store/spl44ld2dp295p5y2s5c63lmaciz3x5l-harfbuzz-7.3.0/lib/libharfbuzz.so.0 (0x00007f2616ae4000)
        libatk-1.0.so.0 => /nix/store/dy0am8lkd08ngrqq36jvg5sl709hd2vp-at-spi2-core-2.48.0/lib/libatk-1.0.so.0 (0x00007f2616fd7000)
        libcairo-gobject.so.2 => /nix/store/67lcda9pla0kmqzabpl3nxxq7z6v4rr6-cairo-1.16.0/lib/libcairo-gobject.so.2 (0x00007f261b20a000)
        libxml2.so.2 => /nix/store/56m1a8i1lmjc15ybkaq6vinq7j2977hv-libxml2-2.10.4/lib/libxml2.so.2 (0x00007f261669a000)
        libsqlite3.so.0 => /nix/store/cc6xrbwk9rinln42n1jlhd9qjmkbv6zb-sqlite-3.42.0/lib/libsqlite3.so.0 (0x00007f2616545000)
        libxslt.so.1 => /nix/store/cdgfvvvv6v365alnsn6zcr4l4xhl05hy-libxslt-1.1.37/lib/libxslt.so.1 (0x00007f2616f99000)
        libdl.so.2 => /nix/store/flf14c3ibr83jsa070j25hg5gjapydhl-glibc-2.37-8/lib/libdl.so.2 (0x00007f261b205000)
        liblcms2.so.2 => /nix/store/mlgys9mnh5sza7f2y72l2vw1qy945zak-lcms2-2.15/lib/liblcms2.so.2 (0x00007f2616a7f000)
        libwoff2dec.so.1.0.2 => /nix/store/a1ba7zr2qzdr358lg5h8lgjr8an5fqvg-woff2-1.0.2-lib/lib/libwoff2dec.so.1.0.2 (0x00007f2618e97000)
        libfontconfig.so.1 => /nix/store/gvhgdpg4xx3zxpkyxaxcmxhxz9nm26yq-fontconfig-2.14.0-lib/lib/libfontconfig.so.1 (0x00007f2616f4d000)
        libfreetype.so.6 => /nix/store/rnrl4zcjcfxqmm982364hlpf3m83nx2m-freetype-2.13.0/lib/libfreetype.so.6 (0x00007f2616477000)
        libharfbuzz-icu.so.0 => /nix/store/f3flracjf0zwz8jpkg59613ln2wc3y5s-harfbuzz-icu-7.3.0/lib/libharfbuzz-icu.so.0 (0x00007f2618e92000)
        libgcrypt.so.20 => /nix/store/72hwj6dppab2gh7kfaas027ljdn9y383-libgcrypt-1.10.2/lib/libgcrypt.so.20 (0x00007f2616330000)
        libgpg-error.so.0 => /nix/store/v37i4li10z5z79k9fgkm97i626iyka1s-libgpg-error-1.47/lib/libgpg-error.so.0 (0x00007f2616a57000)
[... cut ...]
        libelf.so.1 => /nix/store/bx6bs3j995hzdbz7almh4lylv9aabk29-elfutils-0.189/lib/libelf.so.1 (0x00007f26139e8000)
        libbrotlienc.so.1 => /nix/store/lkd8wzdvyyppxxvb5332kkcjscbmwxgq-brotli-1.0.9-lib/lib/libbrotlienc.so.1 (0x00007f2613949000)
        libuuid.so.1 => /nix/store/sh2v8sbg0g3nrwh1dhj2ngzyziamqf86-util-linux-minimal-2.39-lib/lib/libuuid.so.1 (0x00007f261393f000)
        libudev.so.1 => /nix/store/8pbr7x6wh765mg43zs0p70gsaavmbbh7-systemd-253.3/lib/libudev.so.1 (0x00007f2613901000)
```

ImUrX, 2023-08-13:
And I found this issue in appimage https://github.com/AppImageCommunity/pkg2appimage/issues/538

wtarreau, 2023-08-13:
Wow that's an impressive list of dependencies, it's actually amazing it can start and work at all with so many moving pieces. But I understand you have little control over what AppImage does, I really didn't know there was this notion of ExcludeList, please just do not waste your more valuable time on this, really, it's not worth it! thanks for checking anyway!

ImUrX, 2023-08-13:
It's fine, leave this issue open. I'm one of the devs that are trying to make SlimeVR work well on Linux so I really get interested in these issues.

wtarreau, 2023-08-13:
OK then just for your statistics I'm on Slackware 15.0.