SlimeVR/SlimeVR-Server #829: When a rotation vector of nan is received, the application needs to be restarted

wigwagwent, 2023-08-21:
So when the imu reports a nan in the rotation it will not go away once the imu starts sending real data again. For example you short the imu or an incorrect calibration gets loaded to the imu it can return nan. When the sever gets this it will break the the preview (no tracker will show) and will break the rotation display (won't update for that tracker). After the tracker starts sending real data (raw rotation does not break) a full reset will not fix the main rotation and it will still display nan. The only way I have found to get rotation from that tracker again is to restart the server.

[image]

wigwagwent, 2023-08-21:
I had this problem developing for an imu, a day later another discord user also had the same problem when using there mpu6050. They found out when you switch out of dev mode it resets the server so it will work again without having to restart the application.

Louka3000, 2023-09-25:
Solved by https://github.com/SlimeVR/SlimeVR-Server/pull/832

closed 2023-09-25 by Louka3000