SlimeVR/SlimeVR-Server #833: Be able to select network interface for slimevr server

ComputerElite, 2023-08-29:
On my Laptop I got 2 active network interfaces. One for internet (over wifi/eithernet) and one for a wifi hotspot over windows. SlimeVR randomly chooses one of those when launching and thus only works from one of them. Ideally SlimeVR could listen to both ips (e. g. 10.42.1.142 and 192.178.137.1) instead of just one of them.

ComputerElite, 2023-08-29:
Update: Seems like it does listen on all interfaces. However only my smart watch was able to connect over windows hotspot. Could that be a bug in the owoTrack android app?

ButterscotchV, 2023-10-08:
This is likely just an issue with your network routing or firewall. SlimeVR listens on all interfaces (`0.0.0.0`) and does not care about the connection source.

ComputerElite, 2023-10-08:
Oh sorry I didn't update you on this. Yeah, it does indeed listen on all interfaces. I was just dumb. Thanks for your help tho

closed 2023-10-08 by ComputerElite