SlimeVR/SlimeVR-Server #886: Simplify Manual Body proportions configuration

Pixel6554, 2023-10-31:
Currently manual body proportions configuration is very cluttered, we have 2 toggles for ratio and precision, and the list for adjustable proportions keeps getting longer.
Could we...
condense/hide proportions irrelevant to users... possibly based on assigned trackers.
Toggles for basic/advanced/expert view?
I suppose settings should be locked behind 2 or 3 sets of allowed settings. 
Basic view would likely be the main adjustable proportions such as chest/waist/hipwidth/upperleg/lowerleg/upperarm/lowerarm/hand
Advanced may add options for head/neck/hip/feet.
Expert may allow options for all the offsets and other nonsense even i dont understand :)
Im now seeing how annoying this may be to implement but something should be done at some point in making manual proportions easier to adjust for people, a basic/advanced view would greatly simplify body proportions configuration for people

Louka3000, 2023-10-31:
I think 3 modes is too much. Just a “simple” and “full” view should be enough if we were to go that route.

ZycaR, 2024-04-25:
@Pixel6554 I'm not sure, but probably this PR is something which will fix some of your suggestions:
https://github.com/SlimeVR/SlimeVR-Server/pull/1009
especially:
- Toggles for basic/advanced/expert view?
- I suppose settings should be locked behind 2 or 3 sets of allowed settings.