SlimeVR/SlimeVR-Server #894: GUI - first page of Setup Wizard shouldn't block Left side menu

TheLeastMost, 2023-11-05:
Setup Wizard occupies the entire window and if you click on it by accident (and accidental clicks are not uncommon when you using controllers) forces you to go to the top right corner and aim for the close button which isn't that big either. So at least the first page should have acess to the left side menu allowing to quickly leave the wizard.

It would also be nice to be able to completely hide the wizard button somewhere in Settings - Interface - Appearance.

[image]

ImUrX, 2023-11-05:
im not against the second feature. But the first is intentional by design, maybe make the left bar appear after the first setup but :sweat: idk

ImUrX, 2023-11-05:
Btw, with the usage of controller. Are you saying via SteamVR or something like that? I should check if I can check the B button for exiting

TheLeastMost, 2023-11-06:
> im not against the second feature. But the first is intentional by design, maybe make the left bar appear after the first setup but 😓 idk
Only first screen tho. As soon as you click the "lets get set up" it can be wide

TheLeastMost, 2023-11-06:
> Btw, with the usage of controller. Are you saying via SteamVR or something like that? I should check if I can check the B button for exiting

Im on VD, AirLink suffers as well. Once again , I (as a default random user) would not look for some escape button, I just want to click to some other menu button after realising i clicked smth wrong but - they all gone.

PR #1152 (2024-10-08): Hide navbar onboarding
Partially fixes #894 by allowing users to hide the onboarding button to prevent accidental clicks

[image]
closed 2024-10-08 by ImUrX