SlimeVR/SlimeVR-Server #916: Fail tracking

masterofhammer, 2024-01-02:
s it me, or is this new update worse? because literally using VR slime my legs do not fit even though the tracker balls are not inside my leg previously it did very well, now I have to fit them otherwise it does not detect the leg and stays stiff, and on top of that when I move my height is the same It doesn't present reality, and I even have to bend down to track it, it doesn't make any sense. Any advice?

ImUrX, 2024-02-06:
Doesn't seem to be an issue of 0.11 as it's been heavily tested now.

closed 2024-02-06 by ImUrX

masterofhammer, 2024-02-06:
**SOLVED**

The problem was that VRchat had to put the person's real height correctly. And then you had to edit the height with OVR Advanced Settings to adjust it with your feet touching the ground. **Bug fixed!**