SlimeVR/SlimeVR-Server #925: [Feature Request] Walking in place support

FrankDatema, 2024-01-24:
I wish i could use my slimes as a walk in place solution in VR Games. Natural locomotion isn't supported as of now, would be awesome if I could use it for walking around in games like SkyrimVR

ImUrX, 2024-02-06:
So using leg movement of your legs as a way of control in the game?

Louka3000, 2024-02-06:
This should just be sending accel via SteamVR to support NaLo.
Related: https://github.com/SlimeVR/SlimeVR-OpenVR-Driver/issues/11

Montvydas, 2024-03-19:
So WalkOVR is pretty much a competitor to SlimeVR, which supports walk in place. So it is possible and considering the SlimeVR current capabilities, already possible to do given the current algorithms. However one does need to think about how does it all fit into place to stay in the long term?

DatanIMU, 2024-05-05:
It seems that we cannot see the technical details related to WalkOVR. There is a Chinese company called Datan, which has a similar architecture to Slimevr, but they have implemented the Locomotion function. Below is their webpage connection. Can you provide relevant technical references?

https://www.bilibili.com/video/BV1J2421K7ce/?spm_id_from=333.999.0.0

ImUrX, 2024-05-12:
Closing as #25 exists already

closed 2024-05-12 by ImUrX