SlimeVR/SlimeVR-Server #949: Stuck on Connecting to the server

Lumince, 2024-02-27:
I performed a reinstall of windows today and I get this error when attempting to run the server.
[image]
I get this when I bring up the console.
[image]

Lumince, 2024-02-27:
[slimevr_rCURRENT.log](https://github.com/SlimeVR/SlimeVR-Server/files/14425384/slimevr_rCURRENT.log)
Here is the whole log file as well.

ImUrX, 2024-02-27:
We need the server's log which should be called something like `log_last_X.log`

Lumince, 2024-02-27:
[log_last_0.log](https://github.com/SlimeVR/SlimeVR-Server/files/14425685/log_last_0.log)
Its quite empty

Lumince, 2024-02-27:
So I think that there is some left over data that slimeVR server since I did not get the initial setup screen when installing the program?
[image]
Is there any way to fully reset the program from the start?

unlogisch04, 2024-02-27:
to remove all settings delete the following folders:
%appdata%\dev.slimevr.SlimeVR
%localappdata%\dev.slimevr.SlimeVR

Lumince, 2024-02-27:
Yea I tried deleting those as well but the issue still remains. When I installed the program onto my reinstall of windows I did not get the initial setup wizard screen. So theres nothing set in the first place

unlogisch04, 2024-02-27:
Make sure you have installed Visual C++ ( https://github.com/SlimeVR/SlimeVR-Installer/issues/43 )
Did you not get any popup about a error, that a port is in use? (Some ppl reported that on windows 11 with some container features enabled)

Lumince, 2024-02-27:
I have that installed and I still get the same behavior. No pop ups. No errors. I havent even changed any settings since I reinstalled windows this morning. It just refuses to show me the setup wizard or do anything but the above errors

ImUrX, 2024-02-27:
You should check your firewall, as the connection refused part is kinda sus, maybe try https://github.com/SlimeVR/SlimeVR-Server/issues/674#issuecomment-1545860650

Lumince, 2024-02-27:
The server isnt even starting. Seems like its either failing to launch or its crashing?

Lumince, 2024-02-28:
The more I look into it. The more it seems like everyone points towards windows being the issue here and not the program?.... If anyone else has any suggestions, that would be great. I got full body tracking and I can't even use it :(

ImUrX, 2024-02-28:
yeah, maybe reinstall windows, sorry qwq, or maybe you are lucky with the following commands
```
sfc /scannow
DISM /Online /Cleanup-Image /CheckHealth
DISM /Online /Cleanup-Image /ScanHealth
DISM /Online /Cleanup-Image /RestoreHealth
```

Lumince, 2024-02-28:
Yeaaaa looks like I will have to reinstall windows AGAIN. Those commands found no corruption at all so something is just screwed up.

closed 2024-03-09 by ImUrX