SlimeVR/SlimeVR-Server #970: Trackers Over Threshold Visibility for VRChat OSC

ZRock35, 2024-03-31:
Trackers sometimes do not register taps well when using the server installed in the Quest, in standalone mode, with VRChat OSC.

Users have to enable developer mode just to see "Trackers Over Threshold" and it would be good to have it more visible since it's commonly needed.

Additionally it may be worth having it default to 2 trackers over threshold for standalone installations.

Louka3000, 2024-03-31:
I don’t see how being standalone would affect acceleration for tap resets…

ZRock35, 2024-04-01:
I've heard about it many times, it's the reason Trackers Over Threshold was added in the first place. It's a common fix, the reason it happens is unclear to me as well. One thing I can say is repeated tapping will activate the resets, if it's at 1 tracker, but it's very inconsistent. 2 Trackers over works well.