SlimeVR/SlimeVR-Server #989: macOS: Java not killed after exiting app

i0ntempest, 2024-04-14:
Running 0.11.0. After opening and quitting the server app, execute `ps -ef | grep java` in terminal:
```
  501  9051     1   0  2:08PM ??         0:03.52 /usr/bin/java -Xmx512M -jar slimevr.jar run
  501  9092  6866   0  2:08PM ttys002    0:00.00 grep --color=auto java
```
You may see that the java process was not killed. This causes the app to complain "startup error: required ports are busy" on the next run.

ImUrX, 2024-04-14:
can you please show server and gui logs

i0ntempest, 2024-04-16:
Where do I get those logs?

ImUrX, 2024-05-07:
sorry forgot to answer `~/Library/Logs/dev.slimevr.SlimeVR`

i0ntempest, 2024-05-07:
[log_last_0.log](https://github.com/SlimeVR/SlimeVR-Server/files/15241138/log_last_0.log)
[slimevr_rCURRENT.log](https://github.com/SlimeVR/SlimeVR-Server/files/15241139/slimevr_rCURRENT.log)
This was collected after after I killed java after quitting the app.

ImUrX, 2024-05-15:
uhm, did you ever close the app? because GUI logs never say it closed...

do you have tray enabled?

ImUrX, 2024-05-15:
probably related to https://github.com/tauri-apps/tauri/issues/9198, i suspect that Command+Q or Quit is passing as a kill command more than an ExitRequested in Tauri

i0ntempest, 2024-05-15:
That's exactly what I just found out. It handles a click on the X button just fine but not Cmd+Q.

ImUrX, 2024-05-15:
We need to intercept `RunEvent::Exit` then

PR #1034 (2024-06-13): Use `Exit` event instead of `ExitRequested`
Fixes #989
closed 2024-06-13 by loucass003