SlimeVR/SlimeVR-Server #990: macOS: JNA incompatible architecture on arm64

i0ntempest, 2024-04-14:
When trying to launch 0.12 rc3, the app will start but will not connect to server. This is caused by slimevr.jar not starting:
```
/usr/bin/java -Xmx512M -jar /Volumes/slimevr\ 1/slimevr.app/Contents/MacOS/slimevr.jar  run
15:39:20 [INFO] Using log folder: /Users/Admin/Library/Logs/dev.slimevr.SlimeVR
15:39:20 [INFO] Running version v0.12.0-rc.3
15:39:20 [INFO] Using config dir: /Users/Admin/Library/Application Support/dev.slimevr.SlimeVR/vrconfig.yml
15:39:20 [INFO] Starting the tracker server on port 6969...
15:39:20 [SEVERE] SLF4J: No SLF4J providers were found.
15:39:20 [SEVERE] SLF4J: Defaulting to no-operation (NOP) logger implementation
15:39:20 [SEVERE] SLF4J: See https://www.slf4j.org/codes.html#noProviders for further details.
15:39:21 [INFO] [WebSocket] Web Socket VR Bridge started on port 21110
15:39:21 [SEVERE] org.hid4java.HidException: Hidapi did not initialise: dlopen(/Users/Admin/Library/Caches/JNA/temp/jna16742166045332538375.tmp, 0x0009): tried: '/Users/Admin/Library/Caches/JNA/temp/jna16742166045332538375.tmp' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/Admin/Library/Caches/JNA/temp/jna16742166045332538375.tmp' (no such file), '/Users/Admin/Library/Caches/JNA/temp/jna16742166045332538375.tmp' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))
15:39:21 [SEVERE] 	at org.hid4java.HidDeviceManager.<init>(HidDeviceManager.java:87)
15:39:21 [SEVERE] 	at org.hid4java.HidServices.<init>(HidServices.java:82)
15:39:21 [SEVERE] 	at org.hid4java.HidManager.getHidServices(HidManager.java:75)
15:39:21 [SEVERE] 	at dev.slimevr.desktop.tracking.trackers.hid.TrackersHID.<init>(TrackersHID.kt:36)
15:39:21 [SEVERE] 	at dev.slimevr.desktop.Main.main(Main.kt:130)
15:39:21 [SEVERE] Caused by: java.lang.UnsatisfiedLinkError: dlopen(/Users/Admin/Library/Caches/JNA/temp/jna16742166045332538375.tmp, 0x0009): tried: '/Users/Admin/Library/Caches/JNA/temp/jna16742166045332538375.tmp' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/Admin/Library/Caches/JNA/temp/jna16742166045332538375.tmp' (no such file), '/Users/Admin/Library/Caches/JNA/temp/jna16742166045332538375.tmp' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))
15:39:21 [SEVERE] 	at com.sun.jna.Native.open(Native Method)
15:39:21 [SEVERE] 	at com.sun.jna.NativeLibrary.loadLibrary(NativeLibrary.java:300)
15:39:21 [SEVERE] 	at com.sun.jna.NativeLibrary.getInstance(NativeLibrary.java:481)
15:39:21 [SEVERE] 	at com.sun.jna.Library$Handler.<init>(Library.java:197)
15:39:21 [SEVERE] 	at com.sun.jna.Native.load(Native.java:618)
15:39:21 [SEVERE] 	at com.sun.jna.Native.load(Native.java:592)
15:39:21 [SEVERE] 	at org.hid4java.jna.HidrawHidApiLibrary.<clinit>(HidrawHidApiLibrary.java:36)
15:39:21 [SEVERE] 	at org.hid4java.jna.HidApi.init(HidApi.java:105)
15:39:21 [SEVERE] 	at org.hid4java.
```
Apparently updating JNA to 5.7.0 will fix this. See https://github.com/java-native-access/jna/blob/master/CHANGES.md#release-570

ImUrX, 2024-04-14:
it's not an issue in our JNA though... we need to update hid4java

ImUrX, 2024-04-14:
I'm not sure if this will be fixed for 0.12, but there is gonna be a patch release for it if it doesn't get fixed

PR #992 (2024-04-21): update hid4java to 0.8
Fixes #990
closed 2024-04-21 by ImUrX