SlimeVR/SlimeVR-Server #1006: Acceleration decoding doesn't respect AXES_OFFSET

9il, 2024-04-24:
SlimeVR do convert rotation from tracker to openGL AXIS, but forget to do so for acceleration vector.

Louka3000, 2024-04-24:
It does: https://github.com/SlimeVR/SlimeVR-Server/blob/24f28952385dc2f834c62c7157b293ce7613ac50/server/core/src/main/java/dev/slimevr/tracking/trackers/udp/TrackersUDPServer.kt#L357

9il, 2024-04-24:
`AXES_OFFSET` Transforms is a quaternion transformation; it can't make a reflection transformation. It changes Y and Z with a single sign change. 

> It does:

It is reflection transformation (conversion between left-hand and right-hand axis).

9il, 2024-04-24:
@Louka3000 What is the tracker axis in the UDP packet?
 - Z up or Y up?
 - Is it right-handed or left-handed?

Louka3000, 2024-04-24:
I believe SlimeVR expects Y+ forward, X+ right, Z+ up, but not entirely sure right now
OpenVR is Z- Forward, X+ Right, Y+ Up

9il, 2024-04-24:
Understood. We're currently dealing with the Z+ up vs. Y+ Up and hand switch issue.

However,

[SlimeVR-Server/server/core/src/main/java/dev/slimevr/tracking/trackers/udp/TrackersUDPServer.kt](https://github.com/SlimeVR/SlimeVR-Server/blob/24f28952385dc2f834c62c7157b293ce7613ac50/server/core/src/main/java/dev/slimevr/tracking/trackers/udp/TrackersUDPServer.kt#L357)

doesn't affect Z at all.