SlimeVR/SlimeVR-Server #1017: Disappearance of the lower part of the skeleton when there are only thigh trackers

Makkarob, 2024-04-30:
If connect only thigh trackers, the lower part of the skeleton disappears (a full reset does not help), if add foot trackers, nothing changes (their rotation is displayed in the preview).

It is clear that slimeVR should not be used in this way, but the complete disappearance of the lower part of the skeleton, which can only be solved by restarting the program, is also not good.

[image]

ImUrX, 2024-04-30:
someone else has shown me this issue, after deleting their server configuration their legs were back. I asked for their old config and I couldn't replicate it.

Minimewatson99, 2024-04-30:
I had that issue. confused me a lot
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Deleting the vrconfig.yaml fixed it, but I need to do further testing, as the fix wasn't explainable. If I find a direct cause, I will inform you.

ImUrX, 2024-04-30:
thank you, be free to post the actual issue in here!

kruemmelbande, 2024-04-30:
I think ive seen that before. Deleting the config fixed it iirc

loucass003, 2025-09-24:
No new reports of this. Closing