SlimeVR/SlimeVR-Server #1066: Make the tracker mounting radial menu show the current selected direction

loucass003, 2024-06-14:
Currently the radial menu only let you pick a new one, we should also show what it is currently

[image]

ZycaR, 2024-06-17:
Pull Request to fix this issue: https://github.com/SlimeVR/SlimeVR-Server/pull/1077
[image]

closed 2024-06-18 by loucass003

ZycaR, 2024-08-12:
**Updated solution:**
Show the tracker mounting orientation on the manual mounting page for a specific body part only when all mounting orientations of the assigned trackers are the same.

**Steps to reproduce for the case to have two or more trackers on the same body part:**
- Activate the first tracker, assign it to a body part, and select a mounting orientation (e.g., Chest and Front-Left).
- Turn off the first tracker.
 - Activate the second tracker, assign it to the same body part, and select a different mounting orientation (e.g., Chest and Front-Right).
- Turn on the first tracker.

**Result:**
[image]

PR #1077 (2024-06-18): Issue#1066 - Show the current selected direction on the tracker mounting radial menu
Fixes #1066 - Make the tracker mounting radial menu show the current selected direction

The tracker mounting radial menu does not display the currently selected direction, making it difficult for users to know their current selection.

**UI Change**: The selected direction is shown in "dark/black" to ensure it is visible yet not obstructive to the selection of a new direction:
[image]