SlimeVR/SlimeVR-Server #1071: Allow Multiple VRM models to be loaded and selected from in VMC section of SlimeVR settings

ZRock35, 2024-06-14:
Being able to select from multiple VRM model options within the SlimeVR server would be very helpful feature to add - maybe have a reasonable limit.

ImUrX, 2024-06-14:
whats the difference between having to load the file by hand and having the option in the settings, you have to do similar amounts of effort (that is, going to settings)

ZRock35, 2024-06-15:
For quickly switching, without the need to interact with the OS to get to the folder(s) where the VRMs are to access them (drag and drop doesn't work currently, so it's digging through directories a bit, and as nice as it would be to have them in one folder, it's not how I organize my VRMs - those are each with the main blend files etc so I can easily alter them)

All the VTubing programs have that capability, and it would be nice to have some kind of parity, in my opinion, if it's not hard to do - I personally have 4 different VRM models, and I'm just a dabbler in this.

arikie, 2024-07-31:
Hi I use slimevr for mocap in warudo and the ability to use multiple VRMs would be useful because I swap avatars often with automatic chat controlled redeems.