SlimeVR/SlimeVR-Server #1078: Make tracker assigment list use a dropdown on mobile

loucass003, 2024-06-18:
Remove the tracker list on mobile and replace it with a dropdown

[image]

maybe @ZycaR would want to iterate on it ;)

ZycaR, 2024-06-22:
@loucass003 I'm very sorry for the delay (I've been busy at work 😶).
Here is my 'late' proposal, which I have been working on throughout the week. Thank you for understanding!

loucass003, 2024-06-22:
Hi @ZycaR, sorry about all that 🙏 There was another issue related to that new assignment system that i needed to fix so i took the opportunity to do both at the same time. i did it yesterday afternoon and i didnt think you had time to look at it yet. 

It looks like your solution was not so far from mine either ^^ i only splited the component a bit more because i needed to reuse it on the single tracker assignment page.

Please next time do not hesitate to create a draft pr if you take more than a few days to make a PR, that way everybody can see that you are working on it ;)

P.S: idk if you use discord much but we added you in the contributor group, you can now see a lot of the stuff we are working on ;)

ZycaR, 2024-06-22:
Thank you for letting me know. I'll definitely create a draft PR next time (I was just trying to avoid cluttering the PRs, as too many can make the repo a bit messy 😉). 
I'll also check Discord more regularly. Thanks for the update and for adding me to the discord group!

closed 2024-06-26 by loucass003