SlimeVR/SlimeVR-Server #1079: Add a background or a border to the 3d preview so it does not merge with the page background

loucass003, 2024-06-18:
[image]

ZycaR, 2024-07-11:
@loucass003, which background is better: darker or lighter?
[image]
[image]

Additionally, we have a skeleton image on Home page (and other places). Should we apply a background for the 3D preview everywhere, or just on the Manual Body Proposition page?
[image]

loucass003, 2024-07-11:
imo lighter fits the UI more

closed 2024-07-12 by ImUrX