SlimeVR/SlimeVR-Server #1106: Stop allowing people to set any port (osc or vmc) to the tracker port

kruemmelbande, 2024-07-11:
I see people time and time again messing around with the ports and setting for example the osc input to 6969. I think the gui shouldnt allow you to set any port to the tracker port, since doing so will prevent trackers from connecting.

ZRock35, 2024-07-11:
I use other ports to route things for VTubing and Mocap purposes routinely. If anything a warning might be in order, but disabling the ability to set ports is a no-go for me.

kruemmelbande, 2024-07-12:
no, id still want the port to be settable, just not to the tracker port, since setting any port to be the tracker port makes the server stop accepting any trackers at all. A warning might be fine though. If there really is a need to route something through port 6969, you could still change the tracker port in the config, but i guess just adding a warning might be fine too.