SlimeVR/SlimeVR-Server #1112: Battery percentage should not be shown when charging

Vyolex, 2024-07-16:
I've seen a lot of new users get confused by the UI showing 100% charge while the tracker is being charged, thinking it is fully charged regardless of what the actual battery percentage is at.

I feel like the UI should just show a charging indicator and no percentages (that are meaningless and incorrect anyway) as it would be less confusing for new users. Would be nice if later revisions can have current sensing to more accurately display the charge level when charging.

Related to #669 for more clear/better battery reporting.

kruemmelbande, 2024-07-16:
For the most part the tracker doesnt know if its charging. It literally cant tell the difference between being at 100% and being on the charger

Vyolex, 2024-07-16:
Correct, still the server displays a charging indicator which is at >4.3V atm I believe. 

There are cases where it gets mistakenly displayed as charging while it is just fully charged (I have two official dev trackes that do so). I still believe it will be less confusing to remove the percentage compared to what is happening now. But maybe we need to fine-tune the charging threshold value too?

Minimewatson99, 2024-07-16:
my diy slimes say 84% when charging anyways, so getting a consistent result that works across all trackers, is somewhat impossible anyways. 
An improvement could be made for if the tracker is above a certain voltage, but it wouldn't work across all slimes. the usual voltage of a full battery is around 4.2v, and my official slimes read 4.5 when charging, so you could potentially hide battery percentage when the tracker is above 4.3v, which would probably be what you are after. 

Also, because the way the battery percentage is worked out, the percentage of the tracker while charging is impossible to get, as the tracker simply reports the voltage of the battery, and if it is above the maximum voltage that the battery should give, it reports it as charging, so unless estimating the charging time of a tracker, the percentage while charging is not possible to get with the current design of trackers

kruemmelbande, 2024-07-16:
i literally get the exact same value when my tracker is empty but charging and fully charged. The hardware simply doesnt allow you to tell reliably if you are charging or not. If you lower the threashold, itll just start reporting as charging when its not, or itll be inconsistent.

Minimewatson99, 2024-07-16:
> Correct, still the server displays a charging indicator which is at >4.3V atm I believe.
>
> There are cases where it gets mistakenly displayed as charging while it is just fully charged (I have two [...]

fine tuning the charging threshold is difficult, as LiPo batteries vary massively. The best way to get accurate charging status is to go for the lowest voltage that is found while charging different trackers.

Erimelowo, 2024-07-16:
How about just listening for jumps in voltage?  

If a tracker goes from 3.8v to 4.3 (or whatever else) in a second, it’s probably because it got plugged in.

kruemmelbande, 2024-07-16:
I mean, i could see that working while its on, but if you plug it in while its off and then turn it on, there wouldnt really be any way to tell...

Vyolex, 2024-07-16:
Just to be clear since my original message might not be explicit enough, the original intention of this issue is not to do a big reactor of the battery reporting logic (yet?). Instead it's just a request for a small UX improvement, to omit the battery percentage when the server displays the charging icon for a tracker. Currently it will show 100% alongside the charging icon, seemingly throwing off new users, making them believe it's fully charged when it is not.

loucass003, 2025-09-24:
Implemented

closed 2025-09-24 by loucass003