SlimeVR/SlimeVR-Server #1124: Hip snaps around in some specific angles

kruemmelbande, 2024-07-26:
Sometimes the hip snaps around at odd angles. Here i am reproducing it by just moving the hip tracker by hand, although one report suggests that this is happening when sitting down/laying down as well.

Original report in game:
https://github.com/user-attachments/assets/7bf82317-67cd-4a22-807b-0bb1a56e4c6c

My reproduction:
https://github.com/user-attachments/assets/9de00966-6ed0-45b6-8bc0-aa987ba57736

JovannMC, 2024-07-26:
I can confirm I have this issue as well, just had it again today. I also experience it sometimes when I sit/lay - one session I had in VR had me being literally unable to lay down because my body would snap upwards, where I ended up finding it was the hip tracker causing it and disconnected it.

I thought it was something to do with my setup (I'm using the HaritoraX trackers via SlimeTora), but glad to see this isn't just an issue with me.

kruemmelbande, 2024-07-26:
Hmm, weird. Ive seen this happen just for me for a little bit, but i never thought much of it until i heard somebody complain in the discord about it actually happening during gameplay... i wonder how many people actually have something like that...

Erimelowo, 2024-07-26:
That's due to interpolation. Normally, it should allow 270 degrees foward and 90 degrees backwards. This issue can be solved completely by properly keeping track of rotation, which I'm doing in the `fingers` branch...