SlimeVR/SlimeVR-Server #1140: TV UI issues

Minimewatson99, 2024-08-18:
Some smol issues with the UI when running the slimevr server on FireTV sticks and Android TVs shown in the images attached. These issues are not too important, as not many people are likely to run the server on these devices. 
[image]
[image]
More images may be added at a later date

ThomasSteve83, 2024-08-31:
this is probably a common problem problem with the default settings on most tvs today, not even the smart tv device (fire tv stick, etc)

this is usually easy to fix by changing the tv's settings
could also be mitigated by adding some margin space in the root of the ui

not really worth addressing at the server ui level imo

Erimelowo, 2024-08-31:
no, this rather looks like the image was shifted up... If the image was zoomed, the sides would be cut off and the bottom wouldn't be weird like it is.

Minimewatson99, 2024-09-06:
yeah, its not an issue from the TV cropping. i have had this same issue on an android tv, a fire tv stick, and a raspberry pi running android TV

ThomasSteve83, 2024-09-06:
> yeah, its not an issue from the TV cropping. i have had this same issue on an android tv, a fire tv stick, and a raspberry pi running android TV

isn't the common denominator of all those things a tv?
tvs will do this for all input sources (hdmi, etc), not just whatever the tv renders itself

i'm not exactly an expert, but i'm pretty sure nothing about the server gui itself should be responsible for this

ThomasSteve83, 2024-09-06:
> isn't the common denominator of all those things a tv? tvs will do this for all input sources (hdmi, etc), not just whatever the tv renders itself
>
> i'm not exactly an expert, but i'm pretty sure [...]

ohh, i see now

apparently i'm not very good at looking at pictures

knowing how well the server gui runs on everything else, and how terrible most smarttvs are, i'm still unsure if this is a problem with the ui itself, there's a good chance it is

(sorry for being an stupid)

Minimewatson99, 2024-09-07:
I believe it might be related to the way the GUI has to be navigated. Unlike a touchscreen tablet or  phone, or a pc, there is no mouse, so you have to move a selecting box between any objects.
I am going to try and replicate it on a tablet if I can, using an xbox controller