SlimeVR/SlimeVR-Server #1159: Full reset is direction sensitive when using WebSocket HMD

rcelyte, 2024-09-15:
Clicking 'Full Reset' when not perfectly aligned with identity rotation causes all trackers to be offset by some amount. I believe this is due to the WebSocket bridge not setting the flags `needsReset = true` and `isHmd = true` for the HMD tracker like the SteamVR bridge does.

Here's another user's demonstration using the in-development Monado SolarXR driver:

https://github.com/user-attachments/assets/3ec28516-df88-4d34-a723-bee17dcc1f35

yayameowmeow, 2024-09-15:
Maybe slightly better looking video :3, gotta look presentable if i'm gonna be the model for the issue.

https://github.com/user-attachments/assets/35318ce9-5afe-4fa5-b966-d017df998c65

rcelyte, 2024-10-27:
Looks like this was already fixed with #1150 :P. Apologies for the noise.

closed 2024-10-27 by rcelyte