SlimeVR/SlimeVR-Server #1201: Cannot Set WIFI SSID that Contains Emoji

erichowey, 2024-10-15:
Using SlimeVR server, I'm unable to set the wifi when the network's SSID contains emoji. The SSID that is sent is `?` rather than the emoji. This works fine using the Butterscotch firmware tool so it appears to be a SlimeVR Server specific issue.

TheDevMinerTV, 2024-10-15:
I... what.

1. To use emojis, you need proper UTF-8 support which the tracker firmware doesn't have (and will ever have, it's just too big for the small micro controllers)
2. The `?` is very likely because somewhere the emoji's UTF-8 encoding gets broken
3. Why do you have an emoji in your WiFi password... I'm surprised your router is even letting you do that. WPA2 explicitly states that only printable ASCII characters are allowed. (WPA3 isn't supported by the tracker hardware.)

erichowey, 2024-10-15:
> To use emojis, you need proper UTF-8 support which the tracker firmware doesn't have (and will ever have, it's just too big for the small micro controllers)

You might want to double check if you're correct or not. My wifi configuration with the SSID containing emojis works perfectly fine using the butterscotch site or using PlatformIO directly. It only doesn't work with SlimeVR server.

> Why do you have an emoji in your WiFi password...

I'm not. I said i'm using emoji in the wifi SSID - not password

erichowey, 2024-10-15:
See? Works perfectly fine.
[image]

TheDevMinerTV, 2024-10-15:
> I'm not. I said i'm using emoji in the wifi SSID - not password

Oops, missed that. Yeah, the SSID can be whatever, as long as it's 32 bytes with a NUL byte at the end.

Still definitely a weird issue lol

ButterscotchV, 2024-12-08:
Marked as GUI, server, and hardware protocol, as I do not know which step this issue originates from. Please update labels when known.

unlogisch04, 2024-12-08:
Trackers from version 0.4.0 supports bwifi (base64) encoding for the transfer the ssid and of the password. So trackers with version < 0.4.0 do not support that feature.
Transfering utf8 over terminal directly might work, as long there are no null or termination (lf). but did not test that.