SlimeVR/SlimeVR-Server #1252: Add a simple but very useful feature

MLFK8871355, 2024-11-30:
Is it possible to add an automatic tapping function with a timer in the "Tap to Reset"? This feature is really necessary for many scenarios and I think it's quite useful.

JovannMC, 2024-11-30:
Sorry can you explain more of what you're asking, I don't understand what you're trying to say?

MLFK8871355, 2024-11-30:
Sorry, it may be due to the use of machine translation that the translation is inaccurate. What I mean is that in many scenarios where large movements are not required, the drift of the equipment needs to be manually corrected repeatedly. I hope to add an automatic timing device in the software for device calibration, which can automatically time the calibration and save a lot of manual calibration steps

Erimelowo, 2024-11-30:
https://github.com/SlimeVR/SlimeVR-Server/pull/1243

ButterscotchV, 2024-12-08:
Unsure what this means. It's likely this will not be needed after the PR Erimel mentioned. Please re-open if this is not resolved.

closed 2024-12-08 by ButterscotchV

MLFK8871355, 2024-12-08:
I just feel that adding a timer to automatically trigger calibration seems to be very useful in many scenarios, such as myself. Forgetting calibration often causes embarrassment, so I want to have a timer that can automatically calibrate. After all, many times I don't have any other movements, such as my sitting or lying position. Even if I go to sleep with a tracker, it may not calibrate in time and become distorted. Regarding chest tracking, it does solve the problem perfectly, but leg tracking still needs to be manually tapped for calibration. So, what I mean is that in tapping calibration, adding a switch timer that can automatically calibrate after a fixed time is more convenient

ButterscotchV, 2024-12-08:
> I just feel that adding a timer to automatically trigger calibration seems to be very useful in many scenarios, such as myself. Forgetting calibration often causes embarrassment, so I want to have a [...]

The problem is that the calibrations need you to take a specific position, so it would not be possible to just run the calibration. The PR I linked is being worked on further to include leg correction.

MLFK8871355, 2024-12-08:
> > 我只是觉得增加一个定时器来自动触发校准似乎在很多场景下都很有用，比如我自己。忘记校准经常会造成尴尬，所以我想有一个可以自动校准的定时器。毕竟很多时候我没有其他动作，比如我的坐姿或躺姿。即使我戴着追踪器睡觉，它也可能无法及时校准并变得扭曲。关于胸部追踪，它确实完美地解决了问题，但腿部追踪仍然需要手动敲击进行校准。所以，我的意思是在敲击校准中，增加一个可以在固定时间后自动校准的开关定时器更方便
> 
> 问题是校准需要你采取特定姿势，因此不可能在用户不知情的情况下进行校准。我链接的 PR 正在进一步完善，以包括腿部校正。

Thank you very much for your effort. I know this. When I gave the example, it was often a state of not exercising significantly, such as sitting in a chair or lying in bed for a long time, and not exercising for a long time. That's why I had this idea