SlimeVR/SlimeVR-Server #1257: owoTrack connections are not tracked properly by TrackersUDPServer

ButterscotchV, 2024-12-08:
This is an issue between owoTrack reverting to an older protocol without a MAC address causing issues with connection tracking. This can be resolved in the server code.

There is also the issue of the "multi-server" code (new tracker popup) blocking owoTrack connecting, causing it to revert and then it never actually saves properly with the MAC. This can cause tracker duplication, failed connections, and odd tracker timeouts.

There is more to this issue but I no longer remember the details, so I will update this at a later date.

Eirenliel, 2025-03-04:
Maybe owoTrack app needs to be updated?

ButterscotchV, 2025-03-04:
> Maybe owoTrack app needs to be updated?

It probably should, though we may also want to handle this better ourselves

Eirenliel, 2025-03-04:
Probably after https://github.com/SlimeVR/SlimeVR-Server/issues/1326