SlimeVR/SlimeVR-Server #1258: Release v0.14.0

ButterscotchV, 2024-12-08:
Tracking release v0.14.0. 🐄
Estimated release: January-February?

Needs Review(s)/Merge:
- TODO

Merged:
- #1237
- #1240
- #1091
- #1227
- #1235
- #1244
- #1246
- #1242

RC Testing:
1. TODO

ButterscotchV, 2024-12-09:
> I would like to target it end of December. At least have a RC...
> 
> And I really want the new onboarding too.

We can try for December RC, I have a feeling we won't be able to get past RC for December though

ImUrX, 2024-12-10:
Nah, i dont mind either of them, just less burden on having to edit the page when we merge imo

ImUrX, 2024-12-10:
https://github.com/SlimeVR/SlimeVR-Server/pull/1234 requires merge of firmware first also, i think thats how we should merge firmware-dependent PRs for now

FluffyOMC, 2024-12-21:
List needs updated for #1269 merge

ButterscotchV, 2025-04-11:
v0.14.0 is released :3

closed 2025-04-11 by ButterscotchV