SlimeVR/SlimeVR-Server #1263: SlimeVR-Server does not run on Arch Linux with Wayland and nvidia-open drivers

quill-g, 2024-12-15:
**Issue:**
The server application does not run properly and either closes instantly or displays a blank window.

**Steps to reproduce:**
With Wayland:
- install the AUR `slimevr-beta-bin`
- run `slimevr`
- `Error 71 (Protocol error) dispatching to Wayland display.`
- program closes

[image]

Bypassing Wayland:
- run `WAYLAND_DISPLAY=0 slimevr`
- `ERROR [slimevr] Java server receiver died`
- `Failed to create GBM buffer of size 1289x709: Invalid argument`
- attempts to reconnect to webview but never recovers.

[image]

**OS and hardware:** 
Arch Linux x86_64 running KDE Plasma 6.2.4 with Wayland.
Ryzen 7 5800X and GeForce RTX 3080 with `nvidia-open` drivers.

ImUrX, 2024-12-15:
Failed to create GBM is easy to solve, just try one of the following https://github.com/tauri-apps/tauri/issues/9394

ImUrX, 2024-12-15:
aside from that, this is an upstream issue

quill-g, 2024-12-15:
Adding `WEBKIT_DISABLE_DMABUF_RENDERER=1` got it running! Thank you!

closed 2024-12-15 by quill-g