SlimeVR/SlimeVR-Server #1267: Mounting reset causes chest + hip tracker to freeze and disconnect/time out

Twiluff, 2024-12-17:
I noticed that my chest and hip tracker weren’t tracking so I tried to do a mounting reset to see if it would fix the problem. But it didn’t, instead, it froze the chest and hip trackers in the position I was in during the mounting reset, my lower body still works just the upper body that’s broken. It’s strange because it’s only when I do a mounting calibration and the tracker will work absolutely fine if I don’t do it.

KayzaCods, 2025-01-27:
i'm having the exact same problem with my trackers, the hip freezes and gives me [FATAL] [BNO080Sensor:0] Can't connect to BNO085 at address 0x4a, have you solved this issue?

Twiluff, 2025-01-27:
> i'm having the exact same problem with my trackers, the hip freezes and gives me [FATAL] [BNO080Sensor:0] Can't connect to BNO085 at address 0x4a, have you solved this issue?

To be honest, I just went ahead and bought a replacement tracker extension wire from the official SlimeVR store and that fixed the issue. If you have any extensions, you can use a different wire with the Hip tracker to see if it is the wire that is the issue not the tracker, that way you don’t buy a replacement extension wire thinking that it is just the wire and it ends up not fixing the issue.

Twiluff, 2025-01-27:
Oops didn’t mean to do that

closed 2025-01-27 by Twiluff

KayzaCods, 2025-01-27:
I changed the tracker extension from other body part to the chest and also connected the wire upside down and didn't work, factory reset too, so neither is the tracker nor the extension,  may I ask how you got the wire replacement?