SlimeVR/SlimeVR-Server #1273: Won't run/compile from source, couldn't find jdk version error

DoomSlinger, 2024-12-22:
Tried building from source, got this error when running ./gradlew run:

`FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':server:android'.
> Failed to calculate the value of task ':server:android:compileDebugJavaWithJavac' property 'javaCompiler'.
   > Cannot find a Java installation on your machine matching this tasks requirements: {languageVersion=17, vendor=any vendor, implementation=vendor-specific} for LINUX on x86_64.
      > No locally installed toolchains match and toolchain download repositories have not been configured.

* Try:
> Learn more about toolchain auto-detection at https://docs.gradle.org/8.10.1/userguide/toolchains.html#sec:auto_detection.
> Learn more about toolchain repositories at https://docs.gradle.org/8.10.1/userguide/toolchains.html#sub:download_repositories.
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

For more on this, please refer to https://docs.gradle.org/8.10.1/userguide/command_line_interface.html#sec:command_line_warnings in the Gradle documentation.

BUILD FAILED in 378ms
`

Same issue occurs when installing from the AUR: https://aur.archlinux.org/packages/slimevr-server-git

DoomSlinger, 2024-12-22:
> What happens if you just do `./gradlew :server:desktop:run`
> 
> anyways not enough context, seems like gradle can’t find a compatible java version to build with :p

Same error occurs when I use that command. Not sure why it thinks java isn't installed.

ButterscotchV, 2025-06-30:
Please re-open if this is still an issue, it seems more like a Gradle issue than SlimeVR.