SlimeVR/SlimeVR-Server #1280: **minimize to system tray** doesn't work with native window decorations on windows

ThomasSteve83, 2025-01-12:
the server shuts down if you close the gui window:

- while using native window decorations
- from the taskbar
- or in any way other than the gui's close button

this implies that the run in background behaviour is tied to the gui, which is kinda cringe

kruemmelbande, 2025-01-12:
yeah, its called *minimize* to system tray, not *close* to system tray. You press the close button when you wanna.. close the server?

ThomasSteve83, 2025-01-12:
i'm not so sure about that

[image]

Minimize to system tray
Lets you **close** the window without closing the SlimeVR Server so you can continue using it without having the GUI bothering you.

kruemmelbande, 2025-01-12:
[image]

kruemmelbande, 2025-01-12:
its called minimize to tray.. It means that when you press the minimize button, instead of minimizing the gui goes and hides in the tray... so the gui is basically closed...

rcelyte, 2025-01-12:
That isn't how minimize to tray is supposed to work. For every other tray app I've used ([qpwgraph](https://gitlab.freedesktop.org/rncbc/qpwgraph), [Audacious](https://audacious-media-player.org/), [blueman](https://blueman-project.github.io/blueman/), etc.), the minimize button minimizes to *taskbar* whereas the close button closes the window while leaving the tray app running.

kruemmelbande, 2025-01-12:
i dont know what to tell you. Its called "Minimize to tray" which means, that it minimizes to tray. Ive seen plenty of other apps that do it like that. If it closed to tray, it would be called "Close to tray". Its called "Minimize to tray" though, so it minimizes to tray

kruemmelbande, 2025-01-12:
[image]
Look, this is qbittorrent, and it allows for minimize to tray, and close to tray. Those are just different things. Its called minimize to tray, so it does just that

Eirenliel, 2025-01-12:
@kruemmelbande i think you're missing the point of the issue. They're saying that if you press the X on the GUI it goes into tray, but if you're using system decorations it closes, which is a UI inconsistency and it's what the option literally describes should not happen.

So one or the other needs to be changed to be consistent, or both.

closed 2025-01-29 by loucass003