SlimeVR/SlimeVR-Server #1306: [Feature Request] Ability to Offset the Rotation of Individual Trackers Using a Controller

Endakid, 2025-02-11:
I’ve noticed that drift usually affects only one SlimeVR tracker at a time. Since controllers have true 6DOF tracking and don’t experience this issue, it would be useful if we could virtually select/grab an individual tracker and manually adjust its rotation using a controller.

This would allow for quick corrections without needing a full reset, and it would work even while lying down. A standard full T-pose reset could clear any applied rotation offsets.

To keep adjustments precise and prevent overcorrection, it might be worth implementing a scaled-down effect—perhaps requiring two full rotations of the controller to rotate the tracker once. In my experience, only minor rotation offsets are needed to make the avatar look correct again.

(Yes i used chat-gpt to help me reword my schizophrenic dyslexic writing it to be in readable)

Erimelowo, 2025-02-11:
VRChat already has a feature for this. 
Open your quick menu, go in the settings, scroll down to the IK settings, and you’ll find it under the advanced settings

Endakid, 2025-02-13:
> VRChat already has a feature for this, tho it won’t actually affect the SlimeVR skeleton but just the emulated tracker.
> 
> Open your quick menu, go in the settings, scroll down to the IK settings, and you’ll find it under the advanced settings

I just tested it out, that's the exact sort of thing I was thinking! Thank you for telling me about it :)

Endakid, 2025-02-13:
> I'm moving this issue to the server, as this tracker "offset" is handled there.

Sorry for putting it in the wrong one. Thanks for moving it :)

Erimelowo, 2025-03-04:
I think this could still be worth implementing on our end as doing it in VRChat will only affect an emulated tracker and not the skeleton (solve a single tracker’s offset but not its effects on the SlimeVR skeleton)

Eirenliel, 2025-03-04:
Then it should be combined with the revival of overlay...

Endakid, 2025-03-10:
> I think this could still be worth implementing on our end as doing it in VRChat will only affect an emulated tracker and not the skeleton (solve a single tracker’s offset but not its effects on the SlimeVR skeleton)

That's a very good point. also, the VRChat one doesn't let you change the legs or hip (its either the legs or hip, i cant remember. Maybe if you offset it enough the Slime app could use the offset data to help with drift calibration