SlimeVR/SlimeVR-Server #1307: Firmware updater in the GUI is flashing extensions where none exist on official trackers

ZRock35, 2025-02-15:
Firmware updater in the GUI, when flashing official tracker from 5.0, produces extensions that don't exist in the server - gif attached to show the result. Restarting the server does not fix it, reflashing is not possible in the server either.

[image]

Polymoria, 2025-02-15:
From Totikari: its official slimes, and official frimeware updater 

[image]

Piranja81, 2025-02-15:
https://github.com/user-attachments/assets/0ce7666c-f030-4ff5-8989-28a5d90d2cf1
I also have problems with the update

Piranja81, 2025-02-15:
I also have problems with the update

https://github.com/user-attachments/assets/dc00c8b6-dc16-439a-b554-fb78e0740f4c

ButterscotchV, 2025-02-15:
I didn't realize y'all were updating to firmware 0.5.3 that just released today, it's probably that, I've marked it as a pre-release instead on the firmware repo for now, hopefully the updater doesn't pull from it anymore.

Eirenliel, 2025-02-15:
is2g if that's I2Cdev scan again being wonky with BNO... i will rewrite this part.

Eirenliel, 2025-02-15:
Also we need to implement staggered rollout of firmware before 0.14.0 is out @loucass003, as we discussed. It would've been disaster if it was not an RC.

Eirenliel, 2025-02-15:
> I also have problems with the Firmware Update
> SilimeVR.-Firmware.Update.mp4

@Piranja81 we are not talking about the web tool, we're talking about the upater that's in the server. Web flasher will not work while you have serial console open on the server - the port is busy.

ButterscotchV, 2025-02-15:
> I didn't realize y'all were updating to firmware 0.5.3 that just released today, it's probably that, I've marked it as a pre-release instead on the firmware repo for now, hopefully the updater doesn't pull from it anymore.

Can confirm it doesn't seem to want to update anymore since being marked as a pre-release and that firmware v0.5.3 is the problem, moving issue to the firmware repo at https://github.com/SlimeVR/SlimeVR-Tracker-ESP/issues/388.

closed 2025-02-15 by ButterscotchV